incub77/solis2mqtt #1: Solis 5G modbus registers

TheAnt1dot3, 2021-12-20:
Hey man,

First of all, thank you for this great project.
I am a fresh owner of a Solis inverter in use with an external battery setup, and i have a home assistant setup i would like to integrate this with.
After wiring everything up, i noticed the registers you are using are for another generation of Solis inverters.
I have a 5G inverter, for which i've found the modbus registers in a PDF. 
To make it easier for people with the same issue, i have attached the file to this post.
[5G Modbus PDF](https://community.openenergymonitor.org/uploads/short-url/kbwIjN8pl8idrCP04UstaxtSmlC.pdf)
This was originally posted on this [URL](https://community.openenergymonitor.org/t/getting-data-from-inverters-via-an-rs485-connection/8377/188) openenergymonitor

Thank you and keep on doing what you're doing.

Stef

closed 2021-12-20 by TheAnt1dot3

NickSutton, 2022-07-26:
Hi Stef,

Could you post your file please?

TheAnt1dot3, 2022-07-26:
Hi @NickSutton,

The file is directly linked on the ´5g modbus pdf´ url.
Let me know if you have issues accessing the content. 

Thanks,

Stef

NickSutton, 2022-08-06:
Hey Stef, thanks, I can read the document, but not 100% sure which registers to use... I'm not getting the correct 'usage' values at all either, I made a post here: https://community.home-assistant.io/t/solis-5g-pv-register-mapping/447558 with my modbus file posted

NickSutton, 2022-08-17:
`- name: dc_voltage_1
  description: DC Voltage 1
  unit: V
  active: true
  modbus:
    register: 33049
    read_type: register
    number_of_decimals: 1
    function_code: 4
    signed: true
  homeassistant:
    device: sensor
    state_class: measurement
    device_class: voltage

- name: dc_amps_1
  description: DC Amps 1
  unit: A
  active: true
  modbus:
    register: 33050
    read_type: register
    number_of_decimals: 1
    function_code: 4
    signed: true
  homeassistant:
    device: sensor
    state_class: measurement
    device_class: current

- name: dc_voltage_2
  description: DC Voltage 2                   
  unit: V
  active: true
  modbus:
    register: 33051
    read_type: register
    number_of_decimals: 1
    function_code: 4
    signed: true
  homeassistant:
    device: sensor
    state_class: measurement
    device_class: voltage

- name: dc_amps_2
  description: DC Amps 2
  unit: A
  active: true
  modbus:
    register: 33052
    read_type: register
    number_of_decimals: 1
    function_code: 4
    signed: true
  homeassistant:
    device: sensor
    state_class: measurement
    device_class: current

- name: active_power
  description: Active Power
  unit: W
  active: true
  modbus:
    register: 33079
    read_type: long
    function_code: 4
    signed: false
  homeassistant:
[... 311 lines cut ...]
#  Battery Total discharge energy = 33165
#- name: batt_total_discharge_energy
#  description: Battery Total Discharge Energy
#  active: true
#  unit:
#  modbus:
#    register: 33165
#    read_type: register
#    number_of_decimals: 0
#    function_code: 3
#    signed: false

#  Battery today discharge energy = 33167
#- name: batt_today_discharge_energy
#  description: Battery today discharge energy
#  active: true
#  unit:
#  modbus:
#    register: 33167
#    read_type: register
#    number_of_decimals: 0
#    function_code: 3
#    signed: false

# https://gist.github.com/markgdev/ce2dbf9002385cbe5a35b81985f9c84a`