incub77/solis2mqtt #3: Wrong path for installing pip3 Requirements

DerWe, 2022-05-20:
Thanks very much for sharing your Project! I appreciate your work.
I have just a litte improvement, what caused an Error during Installation:

From the documentation, setup.sh is intended to be run like "sudo bash /opt/solis2mqtt/setup.sh".
That way, pip3 can not load the requirements.txt from the opt folder, since the working directory is still different.

echo "### Installing requirements ###"
pip3 install -r requirements.txt

needs to be changed to:

echo "### Installing requirements ###"
pip3 install -r /opt/solis2mqtt/requirements.txt

DerWe, 2022-05-20:
The second Thing I noticed: running "pip3 install -r requirements.txt" as the Default User on my Raspberry pi using sudo, only registers the modules for the user pi. This resulting in being able to run the .py file manually, but forcing the service to fail.

To register the modules globally, you have to switch to the elevated command prompt and install it from there.

sudo su
umask 022
pip3 install -r requirements.txt

Afterwards the system service can start up and the program is running.
Just in case someone else is facing this problem, this might be a hint :)

tux-, 2022-05-24:
Thanks! I had the same issues, and sure enough, this fixed it :)

allgrinder, 2022-06-09:
hey,

I think I ran into this exact error:
`### Installation requirements ###
ERROR: Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'
### Creating config file ###
### Creating service user ###`

But what exactly do I have to do now? Don't understand your hint.