incub77/solis2mqtt #6: solis_modbus.yaml

NickSutton, 2022-07-27:
Line 154, my Pi didn't like the commented text after the register array, I simply moved it to a new line and that solved the issue.

`- name: system_datetime
  description: System DateTime
  unit:
  active: true
  modbus:
    register: [3072, 3073, 3074, 3075, 3076, 3077] # [year, month, day, hour, minute, seconds]`

Needs to be 

`- name: system_datetime
  description: System DateTime
  unit:
  active: true
  modbus:
    register: [3072, 3073, 3074, 3075, 3076, 3077]
   # [year, month, day, hour, minute, seconds]`

If anyone has a full 5G registers config they'd like to share I'd be most grateful, I appear to have some unit/precision errors in mine but will work through them this next weekend.

matthijskooijman, 2022-10-05:
I'm not seeing this issue running on an orange pi and a recent Ubuntu version. Maybe this is related to the yaml library version or so? What OS are you on?

chriskearns, 2022-11-23:
@NickSutton I used [this](https://www.scss.tcd.ie/Brian.Coghlan/Elios4you/RS485_MODBUS-Hybrid-BACoghlan-201811228-1854.pdf) document to select the values I wanted to monitor, and the correct register value. 
Note: the register values should be 5 digits and not 4 as you show above.

Here is my solis_modbus.yaml file, so far so good.

finalbillybong, 2022-12-10:
> @NickSutton I used [this](https://www.scss.tcd.ie/Brian.Coghlan/Elios4you/RS485_MODBUS-Hybrid-BACoghlan-201811228-1854.pdf) document to select the values I wanted to monitor, and the correct register value. Note: the register values should be 5 digits and not 4 as you show above.
> 
> Here is my solis_modbus.yaml file, so far so good.
> 
> [solis_modbus.yaml](https://github.com/incub77/solis2mqtt/files/10075895/solis_modbus.txt)

Any chance of your mqtt sensors too?