incub77/solis2mqtt #8: Slave reported illegal data address

vizkoze, 2022-10-25:
Hello,
I've installed solis2mqtt on a raspberry pi with latest lite distribution of debian.
I use a cheap modbus to ttl converter connected to serial uart pins listening on /dev/ttyAMA0

The inverter is a solis RHI-5K-48ES-5G
Now I don't know if this model uses different modbus registers than mentioned in  solis_modbus.yaml and this is the error? 
I already verified my inverter address is set to 1, default baudrate and stopbit.
When starting the program, after a the inverter comes online, i get following error in the log entry:

2022-10-25 07:24:45,102 - root - Unhandled exception:
Traceback (most recent call last):
  File "/opt/solis2mqtt/solis2mqtt.py", line 185, in <module>
    start_up(args.daemon, args.verbose)
  File "/opt/solis2mqtt/solis2mqtt.py", line 180, in start_up
    Solis2Mqtt().main()
  File "/opt/solis2mqtt/solis2mqtt.py", line 137, in main
    value = self.inverter.read_long(entry['modbus']['register'],
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 599, in read_long
    self._generic_command(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1245, in _generic_command
    payload_from_slave = self._perform_command(functioncode, payload_to_slave)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1329, in _perform_command
    payload_from_slave = _extract_payload(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1880, in _extract_payload
    _check_response_slaveerrorcode(response)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 3538, in _check_response_slaveerrorcode
    raise error
minimalmodbus.IllegalRequestError: Slave reported illegal data address

Any clues what causes this error?

p3tecracknell, 2022-10-25:
That's the same issue I got. I assumed it was an issue my end with the
hardware, so if it is a fixable software error, that would be great news!

vizkoze, 2022-10-26:
I've read that there are some inverters that use a different address structure.
The error seems related to requesting an address that the slave doesn't know about or that the data for the requested address is different than asking for.
Would there be a way of scanning the registers and getting the raw result?

NickSutton, 2022-10-26:
Check the commented out code following the timestamp register is on a new line

vizkoze, 2022-10-26:
Thx for your reply.
I edited the commented out code as per instructions. but I still get the error.
Traceback (most recent call last):
  File "/opt/solis2mqtt/solis2mqtt.py", line 185, in <module>
    start_up(args.daemon, args.verbose)
  File "/opt/solis2mqtt/solis2mqtt.py", line 180, in start_up
    Solis2Mqtt().main()
  File "/opt/solis2mqtt/solis2mqtt.py", line 137, in main
    value = self.inverter.read_long(entry['modbus']['register'],
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 599, in read_long
    self._generic_command(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1245, in _generic_command
    payload_from_slave = self._perform_command(functioncode, payload_to_slave)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1329, in _perform_command
    payload_from_slave = _extract_payload(
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 1880, in _extract_payload
    _check_response_slaveerrorcode(response)
  File "/usr/local/lib/python3.9/dist-packages/minimalmodbus.py", line 3538, in _check_response_slaveerrorcode
    raise error
minimalmodbus.IllegalRequestError: Slave reported illegal data address

Is there a way of scanning the available data addresses of the inverter so i can compare with the entries in solis_modbus.yaml?

NickSutton, 2022-10-26:
You'd be better posting your file

vizkoze, 2022-10-29:
This is my config:
- name: active_power
  description: Active Power
  unit: W
  active: true
  modbus:
    register: 3004
    read_type: long
    function_code: 4
    signed: false
  homeassistant:
    device: sensor
    state_class: measurement
    device_class: power
- name: inverter_temp
  description: Inverter temperature
  unit: "°C"
  active: true
  modbus:
    register: 3041
    read_type: register
    function_code: 4
    number_of_decimals: 1
    signed: false
  homeassistant:
    device: sensor
    state_class: measurement
    device_class: temperature
- name: total_power
  description: Inverter total power generation
  unit: kWh
  active: true
  modbus:
    register: 3008
    read_type: long
    function_code: 4
    signed: true
  homeassistant:
    device: sensor
    state_class: total_increasing
    device_class: energy
- name: generation_today
  description: Energy generated today
  unit: kWh
  active: true
  modbus:
    register: 3014
    read_type: register
    function_code: 4
    number_of_decimals: 1
    signed: false
  homeassistant:
    device: sensor
    state_class: total_increasing
    device_class: energy
- name: generation_yesterday
  description: Energy generated yesterday
  unit: kWh
  active: true
  modbus:
    register: 3015
    read_type: register
    function_code: 4
    number_of_decimals: 1
    signed: false
  homeassistant:
    device: sensor
    state_class: total_increasing
    device_class: energy
- name: total_dc_output_power
  description: Total DC output power
  unit: W
  active: true
  modbus:
    register: 3006
    read_type: long
    function_code: 4
    signed: false
  homeassistant:
    device: sensor
    state_class: measurement
    device_class: power
- name: energy_this_month
  description: Energy generated this month
  unit: kWh
  active: true
  modbus:
    register: 3010
    read_type: long
    function_code: 4
    signed: false
  homeassistant:
    device: sensor
    state_class: total_increasing
    device_class: energy
- name: generation_last_month
  description: Energy generated last month
  unit: kWh
  active: true
  modbus:
    register: 3012
    read_type: long
    function_code: 4
    signed: false
  homeassistant:
    device: sensor
    state_class: total_increasing
    device_class: energy
- name: generation_this_year
  description: Energy generated this year
  unit: kWh
  active: true
  modbus:
    register: 3016
    read_type: long
    function_code: 4
    signed: false
  homeassistant:
    device: sensor
    state_class: total_increasing
    device_class: energy
- name: generation_last_year
  description: Energy generated last year
  unit: kWh
  active: true
  modbus:
    register: 3018
    read_type: long
    function_code: 4
    signed: false
  homeassistant:
    device: sensor
    state_class: total_increasing
    device_class: energy
# frequency currently not supported by home-assistant, topic is in discussion
#- name: grid_frequency
#  description: Grid Frequency
#  unit: Hz
#  active: true
#  modbus:
#    register: 3042
#    read_type: register
#    function_code: 4
#    number_of_decimals: 2
#    signed: false
#  homeassistant:
#    device: sensor
#    state_class: measurement
#    device_class: frequency
- name: system_datetime
  description: System DateTime
  unit:
  active: true
  modbus:
    register: [3072, 3073, 3074, 3075, 3076, 3077]
# [year, month, day, hour, minute, seconds]
    read_type: composed_datetime
    function_code: 4
  homeassistant:
    device: sensor
    state_class:
    device_class: timestamp
- name: serial_number
  description: Serial Number
  unit:
  active: false
  modbus:
    register: 3229
    read_type: register
    function_code: 4
    number_of_decimals: 0
    signed: false
  homeassistant:
    device: sensor
    state_class:
    device_class:
- name: power_limitation
  description: Power limitation
  unit: "%"
  active: true
  modbus:
    register: 3051
    read_type: register
    function_code: 3
    write_function_code: 6
    number_of_decimals: 2
    signed: false
  homeassistant:
    device: number
    min: 0
    max: 100
    step: 0.01
    state_class:
    device_class:
- name: on_off
  description: On/Off
  unit:
  active: true
  modbus:
    register: 3006
    read_type: register
    function_code: 3
    write_function_code: 6
    number_of_decimals: 0
    signed: false
  homeassistant:
    device: switch
    payload_on: 190
    payload_off: 222
    state_class:
    device_class:

p3tecracknell, 2022-11-16:
Is there anything I can do to help? This repo looks amazing and I'm happy to support, but having the same issue

chriskearns, 2022-11-23:
@vizkoze I have used [this](https://www.scss.tcd.ie/Brian.Coghlan/Elios4you/RS485_MODBUS-Hybrid-BACoghlan-201811228-1854.pdf) document to select the values I wanted to monitor, and the correct register value. 
Note: the register values should be 5 digits  for RHI-5K-48ES-5G and not 4 as you show above in your config.

vizkoze, 2022-11-24:
thank you for commenting on this topic. I'll try that one day. Currently i had a lot of succes with a homeassistant modbus plugin named homeassistant-solax-modbus. that plugin used 5 digit register values as well.

NickSutton, 2022-11-24:
This thread looks to be a gold mine too: [](https://www.boards.ie/discussion/2058224611/solar-pv-monitoring-automation-thread/p1)